jfpoilpret/fast-arduino-lib #2: Benchmarks

mikaelpatel, 2016-11-30:
@jfpoilpret This was a nice project you have started. I was looking for some benchmarks to support your claims on the README page. Are there any? Or is this just from "experience"?

jfpoilpret, 2016-11-30:
Regarding size, claims on the README page are based on checking assembly dumps.
Today I can tell what worked for me in complete projects for reducing code size:

- don't use global variables (LDS/STS instructions are 4 bytes and may be used in many locations where a global variable isaccessed)
- avoid virtual: each class with virtual methods has a vtable stored in flash (and copied to SRAM I believe), which seems to be 4 bytes + 2 bytes per virtual method
- avoid vectors if you can (each vector will generate several dozen instruction to push and pop registers): of couse, this is easy said but performance generally requires interrupt vectors, so you have to make your choices.
- use templates for all IO, which enforce CBI/SBI use to access one pin
- use smallest, unsigned, types as possible
- force inline in some instances
- avoid switch and use if else constructs instead
- concentrate all pins on one port and make setup in one time, rather than pin per pin.

I am still yet to find other tricks for further size reduction.

jfpoilpret, 2016-11-30:
Note that my purpose is not to make some comparison benchmarks to see which library is the best.
The main reason I quit Cosa was that some projects would not fit in an ATtiny84A and taking a look at the assembly made me think there were better ways to optimize.

mikaelpatel, 2016-11-30:
@jfpoilpret I can only agree with your notes on optimizations. As you already know Cosa has grown considerably since support for ATtiny was introduced and there are many changes that do not favor for these devices. Actually this puts footprint reduce back on the agenda :) 

As you will notice some of the design decisions you take in the beginning of the project will affect scaling. I did several version with templates in the beginning but abandoned them. 

Also I should mention that the Soft::UART design has very limited optimization put into it. It has not even been update for the new GPIO pin support. 

The interesting thing is to see further development for other IO devices and making a clear interface for device drivers especially support for multiple devices on SPI (which is an interesting challenge :). 

Last but not least many of the constructs in Cosa actually need link time optimization to be reduced. If you have been using the Arduino IDE prior to 1.6.10 there was no support for that and the foot print was larger in general compared to building with Cosa command line (makefile). This might explain why I have not observed the footprint increase for ATtiny. 

Keep up the good work!

mikaelpatel, 2016-11-30:
PS: I guess you will be adding some benchmarks?

jfpoilpret, 2016-11-30:
Actually, size benchmarks are already present (but outdated since I have started refactoring using "traits") in examples-data.xlsx file
I did not consider (so far) performing speed benchmarks but I might add this early next year (I still have too much stuff, more urgent, for 2016).

mikaelpatel, 2017-09-16:
@jfpoilpret After nearly a year I recently got some time to circle back to "GPIO". I started off considering some of your observations of the Cosa design style and stepped back to look at the boarder picture of refactoring towards supporting both AVR and SAM architectures. 

This resulted in an Arduino library for GPIO, https://github.com/mikaelpatel/Arduino-GPIO, with 1) no SRAM usage, 2) single instruction operations, 3) very easy to use, 4) possible to handle multiple architectures within the same source code baseline. The spin-offs are a few new libraries for OWI, TWI and SPI and a bunch of drivers. 

If you have the time please take a look. The benchmarks are impressive; up to 100X compared to Arduino Mega.

closed 2017-12-28 by mikaelpatel