jfpoilpret/fast-arduino-lib #3: Copyright violations

mikaelpatel, 2016-11-30:
@jfpoilpret After reading a fair amount of the code I see that much is directly copied from the Cosa repository and then slightly updated, e.g. file name, inheritance/template, etc. In most cases the original Cosa copyright notice has been removed.

jfpoilpret, 2016-11-30:
Hello @mikaelpatel,

I find your claim to go a bit far, probably farther than it should. It is very unfair too.
Saying "_much of it has been directly copied and slightly updated_" is simply **not true**. If you had really read FastArduino code in detail, you would have seen that there is not as much common between Cosa and FastArduino as you say.
My design is just totally different than yours (almost exclusively based on templates). Please note that template design is a HUGE difference with usual OOP inheritance techniques.

Here are the files where you can claim copyright:
-  NRF24L01: this is clear and this is already stated explicitly in the header. I have already updated headers of these files to contain your copyright notice
- errors.hh: I was lazy on that one initially but now it contains only the list of 4 errors needed, which one could find in any Unix errno.h
- main.cpp:  I took it from Cosa. Copyright header updated already.
- utilities.hh: I used the synchronized macro from Cosa. Copyright header updated already.

Now if you believe there is more of Cosa in FastArduino, then let me know but please make your claims detailed enough.

Please also note that in many cases, any Arduino/AVR library will typically:
- offer the same features (those from the MCU)
- share some implementation code (most code directly derives from ATmel datasheet information and can hardly be changed to something much different)

Regards

mikaelpatel, 2016-11-30:
@jfpoilpret I understand that this is very early in the game and you actually have not yet come around to adding a license and a header to most of your files. 

In Cosa I have tried my best to acknowledge algorithms, code sections, inspiration, etc. It helps the community grown and add some history to the development. There is very little new stuff under the sun :). 

And you have several ideas that I think will help evolve Cosa especially with your focus on ATtiny. 

Last but not least I would like to suggest that you add some info that this core as Cosa does not implement the Arduino API. The name is therefore somewhat misleading (fast-arduino-lib). 

Cheers!

jfpoilpret, 2016-11-30:
@mikaelpatel Thanks for your constructive feedback.

You are right, I did not care (at all) about file headers and copyright stuff as I wanted to go very fast:  FastArduino project has taken all of my free time since June, and I completely put on hold my other project "alarm system" until FastArduino gets all that I need for that other project (almost there but not yet, still).

I have updated the README, a couple of minutes ago, and included the fact it does not support Arduino API (and will never, as a matter of fact).

Cheers

jfpoilpret, 2017-02-03:
Licenses have been set across all source files now.
FastArduino now uses ASL2.0 for everything but NRF24L01P that uses LGPL2.1 as derivative work from Cosa.
I have removed the few remaining parts that I had copied from Cosa (synchronized macro implementation, C++ ABI functions).

closed 2017-02-03 by jfpoilpret