jfpoilpret/fast-arduino-lib #28: project status

susnicek, 2018-12-11:
Dear developer,
this library is something I was looking for a long time. What is the project status, is it still alive?
Best regards, Susnicek

jfpoilpret, 2018-12-12:
Dear Susnicek,
I am glad that fastarduino is a library that could fit your needs.
Although I had had no time since May to go on working on this project, it is still alive, be reassured!
I just hope I can go back to work on fastarduino library in January; however, I may reduce the roadmap a bit to allow more frequent releases.
I'll need first to review and replan all open issues; I'll try to do that during the end of year break.
Cheers, Jean-Francois

jfpoilpret, 2018-12-12:
I will comment more here as soon as I have updated the roadmap and started back my work.

susnicek, 2018-12-13:
Dear Jean-Francois,
thank you for your reply! I am looking forward to your news about the project.
Best regards, Susnicek

mgoodings, 2018-12-17:
This project is fricken awesome!

jfpoilpret, 2019-04-12:
Dear Susnicek, I finally found time to restart working on FastArduino.
I have updated issues and milestones.
I have just finished a redesign of ISR handling (heavy changes and simplification of all REGISTER_XXX macros).
Cheers
Jean-Francois

closed 2019-04-12 by jfpoilpret

susnicek, 2019-04-16:
> Komu: jfpoilpret/fast-arduino-lib <fast-arduino-lib@noreply.github.com>
> "

Dear Jean-Francois,

thank you very much! This is perfect!

Best regards, Susnicek

---------- Původní e-mail ----------

(https://github.com/jfpoilpret/fast-arduino-lib/issues/28#issuecomment-482444826)
"