jfpoilpret/fast-arduino-lib #48: Add support for user-selectable cystal frequencies

susnicek, 2019-06-19:
Dear developer,
I would like to ask if it is possible to add a support for user-selectable cystal frequencies. Maybe F_CPU macro will work, I do not know.
My point is to use your library for general code development for processors ATmega, ATtiny and their families. And I do not want to be bound to a specific board platform with pre-soldered crystals etc. Will be your library also aimed this way?

Best regards, Susnicek

jfpoilpret, 2019-06-21:
Actually this is already supported by the library and the make system.

As you guessed, you can set `F_CPU` to the crystal frequency you need.

There is one important constraint with values of `F_CPU` though: they must be multiple of 1 MHz because several expressions in FastArduino code divide `F_CPU` by `1'000'000` for some calculation (timing utilities essentially but not only).

The make system has 2 ways to be configured:
1. CONF=predefined config name
2. VARIANT=target, MCU=mcu, ARCH=target AVR arch, F_CPU=frequency

To achieve what you want, you need to use the second approach.

If you take a look at `Makefile-config.mk` file, you will see what predefined configurations exist, and how they map to individual settings.
What is important to understand is that only the following `VARIANT`s are supported by FastArduino (as described in the README):
- ARDUINO_UNO
- ARDUINO_NANO
- ARDUINO_LEONARDO
- ARDUINO_MEGA
- BREADBOARD_ATMEGA328P
- BREADBOARD_ATTINYX4
- BREADBOARD_ATTINYX5

As long as you stick to these boards and MCUs, FastArduino library shall work, even with different frequencies than the defaut.

Other AVR MCUs are not currently supported by FastArduino because:
- I do not use (nor own) other AVR chips (I decided not to support devices I cannot test)
- Adding new MCU targets to FastArduino is not a straightforward task (I intend to document it but that is not my top priority so far); if you want to have an idea of how it works, you can take a look at `/cores/fastarduino/boards` directory, for example the pair of headers `uno.h` and `uno_traits.h`. 

That second point should show you that FastArduino is aimed to rather easily support more AVR targets.

jfpoilpret, 2019-06-22:
@susnicek please let me know if you need more information on this topic. I'll do my best to answer.
Unfortunately I cannot document yet how to add new MCU targets (I planned it before end 2019, this is issue https://github.com/jfpoilpret/fast-arduino-lib/issues/17).

susnicek, 2019-06-25:
Dear @jfpoilpret ,
thank you very much for your answer! It is fine to have the possibility.
Currently, I have a problem to compile the fastarduino library.
I am going to raise another issue on this problem.

Best regards, Susnicek

jfpoilpret, 2019-07-01:
Question has been answered, issue can be closed.

closed 2019-07-01 by jfpoilpret