jfpoilpret/fast-arduino-lib #49: Library compilation problem

susnicek, 2019-06-25:
Dear @jfpoilpret,

I followed the [documentation](http://jfpoilpret.github.io/fast-arduino-lib/) and  tried to compile the fast-arduino-lib package using the command 'make CONF=UNO build'. However, the command yielded the following error:

`avr-g++ -MT build/ARDUINO_UNO-16MHz/cores/fastarduino/soft_uart.o -MMD -MP -MF deps/ARDUINO_UNO-16MHz/cores/fastarduino/soft_uart.Td -mmcu=atmega328p -DF_CPU=16000000UL -DARDUINO_UNO -DNO_ABI -fno-exceptions -Wextra -flto -felide-constructors -Os -ffunction-sections -fdata-sections -mcall-prologues -g -Wall -I /media/data/Documents/Elektro/FastArduino/fast-arduino-lib-master/cores -std=c++17   -c -o build/ARDUINO_UNO-16MHz/cores/fastarduino/soft_uart.o /media/data/Documents/Elektro/FastArduino/fast-arduino-lib-master/cores/fastarduino/soft_uart.cpp
In file included from /media/data/Documents/Elektro/FastArduino/fast-arduino-lib-master/cores/fastarduino/soft_uart.cpp:15:0:
/media/data/Documents/Elektro/FastArduino/fast-arduino-lib-master/cores/fastarduino/soft_uart.h:67:17: error: expected '{' before '::' token
 namespace serial::soft
                 ^
/media/data/Documents/Elektro/FastArduino/fast-arduino-lib-master/cores/fastarduino/soft_uart.h:67:19: error: 'soft' in namespace '::' does not name a type
 namespace serial::soft
                   ^
/media/data/Documents/Elektro/FastArduino/fast-arduino-lib-master/cores/fastarduino/soft_uart.h:176:39: error: expected '}' before end of line
/media/data/Documents/Elektro/FastArduino/fast-arduino-lib-master/cores/fastarduino/soft_uart.h:176:39: error: expected declaration before end of line
make: *** [/media/data/Documents/Elektro/FastArduino/fast-arduino-lib-master/Makefile-common.mk:131: build/ARDUINO_UNO-16MHz/cores/fastarduino/soft_uart.o] Error 1
`

My compilation attempt was performed under the Debian Linux 4.19.37-5 (Sid), version of my avr-g++ is 5.4.0. Any advice on this issue?

 Best regards, Susnicek

susnicek, 2019-06-25:
Probably, my avr-g++ version is too old...

susnicek, 2019-06-27:
Ok, I used older syntax for nested namespaces in soft_uart.h, line 67

`namespace serial
{
namespace soft
{
...}}`

...and the compilation was successful.

I thought, that the compiler's switch '-std=c++17' would do all magic, even if the 'avr-g++' version is not the latest, but it would not...

Maybe, it is better to use the old (and ugly) syntax, the most users have older compilers - even me with Debian Unstable distribution.

Best regards, Susnicek

jfpoilpret, 2019-06-27:
Hi Suscnicek,

yes your avr-g++ version is pretty old. Nested namespaces (part of C++17 standard) are not supported by g++ version before 6.0.
However, even g++6.0 is not supporting all C++17 standard features and I am not sure 6.0 support would be enough fro FastArduino.

In particular, there is much better support for `constexpr` functions in C++17 which allow simpler definitions of such functions (without such support, it is not possible to have more than one `return` in a `constexpr` function, which makes writing such functions more complex and lessen their readability). 

For what it's worth, FastArduino is built (on my setup) with **avr-g++ 7.4** with various C++17 features (nested namespaces and improved constexpr functions at least), and I don't have plans to support older versions as this is the recipe for bad-looking source code; note that even g++7.4 is not that new actually (released in 2017 I think), but this was the latest version available for AVR targets on fedora distrib, which is my distrib of choice on all my workstations). I generally reserve Debian for servers only (plus Raspberry Pi with raspbian distro).

susnicek, 2019-06-28:
Dear @jfpoilpret ,
thank you for your answer. It is ok for me now, I can use the libraries with the slight modification (or maybe later I will try to compile the avr-gcc sources etc.).
One think is not clear to me: the compiler complained on the nested namespace in 'soft_uart.h' but it did not complain on the same thing in 'uart.h'. I only needed to modify the file 'uart_soft.h' to compile the library. Strange...

Susnicek

jfpoilpret, 2019-06-28:
Hi @susnicek 

I think the main reason why the compiler did not complain for `uart.h` is that it did not compile anything including `uart.h` (headers are not compiled during library building only .cpp source files).

`soft_uart.h` gets parsed by the compiler because it is included in `soft_uart.cpp` which must be compiled.

When you will use (in an application) FastArduino headers, you may get more of this error. In particular, if you use FastArduino headers for **devices** support, nested namespaces are also used there.

So in the end, you will need to change nested namespaces whenever you include headers where they exist.

jfpoilpret, 2019-07-01:
Issue has been answered and can be closed.

closed 2019-07-01 by jfpoilpret

susnicek, 2019-07-02:
Yes, you were right.

Thank you!