jfpoilpret/fast-arduino-lib #52: MPU6050::all_measures swaps accelerometer and gyroscope

urdh, 2019-08-13:
Using the following snippet:

```c++
devices::magneto::Sensor3D accel;
accelgyro.accel_measures(accel);

devices::magneto::Sensor3D gyro;
accelgyro.gyro_measures(gyro);

devices::magneto::AllSensors sensors;
accelgyro.all_measures(sensors);

output << streams::setw(6) << "var"   << " |" << streams::setw(8) << "sep."  << " |" << streams::setw(8) << "comb."         << "\r\n";
output << streams::setw(6) << "ACC_X" << " |" << streams::setw(8) << accel.x << " |" << streams::setw(8) << sensors.accel.x << "\r\n";
output << streams::setw(6) << "ACC_Y" << " |" << streams::setw(8) << accel.y << " |" << streams::setw(8) << sensors.accel.y << "\r\n";
output << streams::setw(6) << "ACC_Z" << " |" << streams::setw(8) << accel.z << " |" << streams::setw(8) << sensors.accel.z << "\r\n";
output << streams::setw(6) << "GYR_X" << " |" << streams::setw(8) << gyro.x  << " |" << streams::setw(8) << sensors.gyro.x  << "\r\n";
output << streams::setw(6) << "GYR_Y" << " |" << streams::setw(8) << gyro.y  << " |" << streams::setw(8) << sensors.gyro.y  << "\r\n";
output << streams::setw(6) << "GYR_Z" << " |" << streams::setw(8) << gyro.z  << " |" << streams::setw(8) << sensors.gyro.z  << "\r\n";
```

I would expect the acceleration and gyroscope measurements to be very similar between these three calls (*i.e.* that `sensors.accel.x` be very close to `accel.x` and so on), but instead I get this:

```
   var |    sep. |   comb.
 ACC_X |    1800 |     -21
 ACC_Y |     -60 |    -164
 ACC_Z |  -16288 |     -20
 GYR_X |     -16 |    1728
 GYR_Y |    -170 |    -148
 GYR_Z |     -30 |  -16364
```

It looks to me like the driver is reading the gryo and acceleration data in the wrong order.

jfpoilpret, 2019-08-14:
Hi Simon, thank you for your report.
I will check this point this week. I guess the fix should be straightforward.
I'll put updates here.
Cheers
Jean-François

jfpoilpret, 2019-08-16:
Hi Simon,
I can confirm this is a bug.
The fix was obvious (sorry, I should have spotted it before).
Fix is in commit https://github.com/jfpoilpret/fast-arduino-lib/commit/74be1182f69262ee2babc2a73da2b97caa849d5a
Thanks a lot for your report and contribution to FastArduino!
Jean-Francois

closed 2019-08-16 by jfpoilpret