jfpoilpret/fast-arduino-lib #73: ArduinoDevSetup.docx link gives 404 error

hreintke, 2020-01-06:
Hi,

The ArduinoDevSetup.docx. link in the readme gives 404 error.
Could not find it on the site either.

jfpoilpret, 2020-01-06:
Hi @hreintke thank you for your report. 
After recent restructuring of FastArduino docs, I completely forgot to update this link.
This is done now as per commit db1dbdb694fae95a4ad4175c54794ba1d2d7b427 on master branch.

jfpoilpret, 2020-01-06:
For your information, several pieces of documentation have been recently moved to directory `basedoc`, in order to reduce the number of files in root directory of the repository.

closed 2020-01-06 by jfpoilpret

hreintke, 2020-01-06:
@jfpoilpret 
Found and read the setup document.
I am on windows, do you know whether
- Anyone has Fastarduino running on windows ?
- The toolchain from https://blog.zakkemble.net/avr-gcc-builds/ can be used for FastArduino ?

jfpoilpret, 2020-01-06:
@hreintke I don't know if anyone has tried (and succeeded) running FastArduino make on Windows.
For what it's worth, it has taken me a lot of time to setup FastArduino make system and I had to use GNU make specificities, which probably makes the system linked to linux, maybe GNU shells on Windows may work (I never tested).
Your best bet is probably to use a Linux Virtual Machine on your Windows box (this is what I did first before definitely moving my PC from Windows to linux; at that time, I used VirtualBox freeware). I would advise Fedora (this is the official distrib for my work everyday and for FastArduino in particular.

The toolchain you mention should work, I think, at least for the compilation part. Not sure for the `make` part itself. But that seems worth trying, I would say.
If you try it, do not hesitate to comment here so that there is a track on how to do that on Windows. I could then create a special setup guide for Windows machines (although I cannot try that by myself since I have no Windows left at home).

hreintke, 2020-01-07:
@jfpoilpret I took another route.

My Arduino/ESP development environment is Eclipse.
For Arduino this contains avr-gcc 7.3.0, made it use -std=gnu++17
Created a new Arduino Nano project
Added part of the FastArduino platform to the project. 
For that I needed some small updates to the FastArduino files (conflicting defines) , adding #define ARDUINO_NANO to some of the files.
First examples run correct. gpio/uart/analog.

For now I will use this setup to see whether using FastArduino or not.
If it works OK, you could consider making FastArduino an Arduino library and in that way having Arduino IDE support implicit.

hreintke, 2020-01-08:
@jfpoilpret 
Must be overlooking something.
What would be the FastArduino equivalent of : 
```
#define ARDUINO_NANO
#include "fastarduino/gpio.h"

#include "Arduino.h"

class Motor
{
public:
  Motor (uint8_t pin) : _pin(pin){};
  ~Motor(){};
  uint8_t _pin;
  void func()
  {
	  // use _pin
  }
};

Motor m1{1};
Motor m2{2};

void setup()
{
}

void loop()
{
	m1.func();
	m2.func();
}
```

jfpoilpret, 2020-01-08:
Hi @hreintke 
As mentioned in FastArduino README, FastArduino is fundamentally different than Arduino API and uses templates to ensure that you cannot build a program that would use features that do not exist on the target of your program. This requires more effort than using Arduino API but it provides safer (and more beautiful, I would say, but I'm not objective :-) ) code. I highly recommend reading the tutorial to understand how to use FastArduino API.

Regarding your sample code, I would first recommend not forcing `#define ARDUINO_NANO` directly in your source files otherwise you'll have to add it everywhere (if your project has many source files) and update it every time you need to change Arduino target. The recommendation is to pass it to the compiler command line. Whatever tool you use to build your program, it probably supports setting defines on the command line calling the compiler.

hreintke, 2020-01-09:
@jfpoilpret 
Don't mixup proof of concept code with final usage. Yes, my development environment uses defines to allow support to different boards, but those are not compatible with yours.
Not difficult to adapt the FastArduino ones to the ArduinoIDE standard but not necessary for the first step. Codewise : For the example above I am looking into a templated Motor class.

In my projects I use code sharing between AVR and ESP. I cannot leave Arduino API totally.
That is why I will cherry pick functionality from FastArduino and integrate that in Arduino.
The work from my side but is probably worth the effort.

Thanks for you effort in FastArduino,  will follow closely and report issues/improvements I see.