jfpoilpret/fast-arduino-lib #74: Add FastDigialPin as type

hreintke, 2020-01-08:
I do not (yet) have a github fork from your repository so this as an issue.

I am not experienced in c++ templates but looking thru your code and a lot of reading :
When you add : 
```
	template<board::DigitalPin DPIN_>
	using FastDigitalPin = typename FastPinType<DPIN_>::TYPE;
	/// @endcond

}
#endif /* FASTIO_HH */
/// @endcond 

```
to the end of `gpio.h` 
You add the functionality to use 
```
     FastDigitalPin<DigitalPin::LED> led{PinMode::FASTOUTPUT};
```
as alternative for 
```
     FastPinType<DigitalPin::LED>::TYPE led2(PinMode::FASTOUTPUT);
```
Leading to more readable an understandable code

jfpoilpret, 2020-01-08:
Hi @hreintke this looks interesting indeed, thank you for this suggestion.
I'll need to check out that this can work in all situations before adding it.
For that, I think the best is to use it in all FastArduino examples and check everything builds OK.

jfpoilpret, 2020-01-08:
I'll work on it for next version 1.5.
I'll also need to:
- add API doc
- update tutorial to include this new API

jfpoilpret, 2020-01-12:
Issue implemented on master branch as per commit https://github.com/jfpoilpret/fast-arduino-lib/commit/f3485e01fdd1179f17fdc16cc8c18c5bd216f148

commit f3485e01 (2020-01-12): Merge pull request #75 from jfpoilpret/improve_FastPin_typing

Fix #74 improve FastPin type declaration
closed 2020-01-12 by jfpoilpret