jfpoilpret/fast-arduino-lib #83: Compatibility with avr libraries

susnicek, 2020-06-11:
Hi @jfpoilpret, 
it would be nice that fast-arduino-libs were compatible with standard avr Libc libraries (e.g. avr/io.h etc.). This compatibility is preserved with arduino libraries (avr headers can be normally used), however, concurrent use of fast-arduino-libs and avr headers ends with errors.

Best regards, S.

jfpoilpret, 2020-06-13:
Hello @susnicek, would you have a concrete example that shows this incompatibility?
Normally, you should be able to develop a program that mixes both, with only one constraint as follows: 

You should not **include** FastArduino headers and AVR libs headers **in the same source code file**.

But you should be able to have, in the same program:
- one cpp file including AVR headers
- another cpp file including FastArduino headers

In this case, compilation and linking should work fine.

If you need to include both headers in the same file, then it will not be possible. 
The main reason for that is the **avr/io.h** header: it is heavily based on macros and dedicated to pure C programs but not optimized for C++; it wass a nightmare to use it directly inside a C++ library like FastArduino, where lots of checks can be performed at compile-time.

susnicek, 2020-06-15:
Hi,
it exactly as you say, a mixed code (inclusion of AVR and FastArduino headers) in one file did not work for me. 

Best regards, S.

jfpoilpret, 2020-06-21:
As mentioned before, this is not something FastArduino can possibly circumvent, otherwise that would mean removing the possibility to have compile-time errors on usage of unexisting features (ports, IO, timers...)

closed 2020-06-21 by jfpoilpret