jfpoilpret/fast-arduino-lib #86: Renaming library name...

Gamadril, 2020-10-08:
Maybe you should think about renaming your library. I was looking for an alternative to arduino libraries for my next project and found your repository. At the first sight it's very promising and I will give it a try. However the name is very confusing, since it has nothing to do with Arduino itself (it's more an AVR library). I know you have a remark in the README about the incompatibility to Arduino libraries and this is another reason to rename the lib...

jfpoilpret, 2020-11-15:
Hi Gamadril, thank you for your comments.

As a matter of fact "Arduino" has many different meanings; this may be understood as:
- a board or set of MCU boards with trademark Arduino
- a set of libraries ("platform") for these boards
- an IDE
- an ecosystem with all the above and plus

In order to name **FastArduino** library, I chose the first meaning (a set of boards).

Actually you are (partly) right in saying FastArduino library is an AVR library. It is correct but it is a library that natively supports Arduino boards (not all, but the most common: UNO, Leonardo, Mega) .
Hence it is what we could call a new "platform" for AVR MCU and AVR-based Arduino boards (including clones by the way).

In order to solve the search issues, I have adapted the keywords of the github project, modified the description and slightly changed the README. I insisted more on the "platform" aspect. I checked several searches and found FastArduino project to show in much better position in the results, and always in the first page.

I am not very keen on changing the library name itself, though. I had a hard time selecting it:
- I wanted a unique name
- I wanted to avoid trademark issues (with Atmel, then Microchip in particular)
- I did not want to rule out Arduino boards users
- I wanted a name easy to understand and remember

With all this in mind, I chose "FastArduino" as a relevant name. Maybe it is not fully satisfactory but, after more than 4 years of existence, I have been "attached" to its name (and several users I know use it "fluently" too).

Now I am still open to suggestions, so if you (and other people following this project) can come up with better names, respecting constraints enumerated above, I will study all proposals with great attention.

For the moment I keep your issue open but I will close it on 30.06.2021 if nothing comes up until then.

Best Regards

Jean-Francois

jfpoilpret, 2021-08-29:
In the absence of better names suggestions to FastArduino library after a long time, I think I can close this issue.

closed 2021-08-29 by jfpoilpret